nunobrum/spicelib #218: RawFile does not read integral noise

simonwaid, 2025-07-14:
When performing a noise simulation in ngspice, at the end of the raw file the Integrated Noise is saved as a separate "Plot".  It would be useful to be able to access it. However, it seems RawFile ignores it. The following contains a .raw file with such a noise simulation. Scroll to the end using any text editor to see the entry. 

[testfile_noise_ss_35.0_3.2_0.zip](https://github.com/user-attachments/files/21219326/testfile_noise_ss_35.0_3.2_0.zip)

hb020, 2025-07-17:
Yes, this is a subject we've stumbled upon a couple of times. The different simulators all behave slightly different here, and not all can write and/or read multi-section raw files, especially since they sometimes use some nasty dialects. 

My recommendation is to avoid multiple sections, to keep compatibility with a large range of tools. 

Our rawread cannot read more than 1 section at this moment. We can however work on that, as this is not the first time we see it. On top of that, we should also better support that second section type you have: "Integrated noise".  Tell us if this is important for you, and I can get it going. Would like some more test files then, both in ASCII and raw, as creating test files with different content types is somewhat of a challenge.

simonwaid, 2025-07-22:
I find the feature of reading multiple sections a very useful feature. Besides reading the output of noise simulations, this would also be helpful with control section and loops, e.g. to change the simulation temperature or perform monte carlo simulations.

Regarding the issue with reading integral noise as an intermediate solution, I've hacked RawRead a bit so it skips reading everything before it finds `Plotname: Integrated Noise`. I found RawRead reads the data just fine. Please find attached a raw and asc file containing the same data. The files were created using a control section and the appendwrite option.

Please let me know if you need more examples, such as stepping temperatures in a loop or similar.

[test.zip](https://github.com/user-attachments/files/21371950/test.zip)

hb020, 2025-07-23:
Well, any combinations would be great. Also, if you have show how you create them, that would be nice, as I rarely use ngspice, so I'd need to some digging first.

hb020, 2025-07-23:
@nunobrum if you are looking, this is about being able to read raw files that have multiple trace sets/plots in them. ngspice and xyce can do that, and maybe also others. When you have multiple trace sets, you simply have concatenations of multiple fully valid raw files into 1 file.

I am planning to adapt the RawRead class with an extra option: "get the Nth plot from the file", with 1 being the default. That way it is fully retrocompatible. 
And yes, I will make sure it support stepped files (and that unit tests have stepped files tests).

nunobrum, 2025-07-25:
I'm now looking. I'll have a bit more time next week to have a deeper look at it. 

I'll also check all the other issues now pending.

nunobrum, 2025-07-27:
@hb020  I had a look at PR #221 and the zip file that @simonwaid sent. I understand now what different plots mean.
First of all thanks for the really nice work you have done there. I love that you have created a dedicated Error Class for managing Exceptions and have greatly improved the unittests for that class.

The implementation looks solid, however, before making the pull, I would like to present some ideas for your consideration.
Then we can discuss if we keep the PR as is, or we update it. I am also open to other ideas. 

In the current implementation, a raw file corresponds to one single Plot (Let's call it that way), hence it made sense to have all the decoding of the RAW file on its constructor.
If we advance to having multiple plots in a single RAW file, then, for me it would be more natural as an interface, that each Plot would be represented by a single class. And in this case, the parsing of the raw file would become rather become a function that would return one object per plot. Let's call this option a).
For backwards compatibility, if a single plot exist on the plot, then a single object is returned, and if there is more than one plot, then, it would return an array of plots. The other options are : 
b) to keep @hb020 approach with having an extra argument
c) to keep one single class that returns the first plot, and the 2nd, 3rd and subsequent would be accessed by a property
d) Same as a) but breaking compatibility of previous implementations by returning always an array

For me the advantage of a) is that it feels more natural for the user. The option b) requires knowing in advance the structure of the raw file, or alternatively, having to ask first how many plots exist on the raw file.

There is another idea that I had a long time ago, but never implemented, was to have a lazy loading of the trace vectors. The idea is to have a constructor/function that only parses the headers, and the data is only read when the user requests for it. The advantage of this is saved memory and speed. In most cases, the user only requires a subset of the data on the file, so, there is no real use of reading everything into memory. This would avoid the usage of dummy waves as it is done now. Also, it would move the wave parsing code from the constructor/function to the Trace Class, which, I believe would make the code a bit more clear.

That's it. As said, I am open to your ideas, and by presenting this, I am by no means imposing anything. Opinions may vary according to different perspectives. My wish is to have the better interface possible given the constraints we have.

hb020, 2025-07-28:
Not easy this one.

tl&dr;

So to me, we should provide option a), but keep a backwards compatible interface. 

Details:

* The interface as it exists is fundamentally based upon 'one plot at a time', because of the public exposition and use of the instance properties (nVariables, nPoints, aliases, backannotations, has_axis, axis). The instance methods are not blockers here,  as they can be made multi-plot compatible.
* backward compatibility is very important here: although the interface can be improved, it is not broken, and caters for at least 99,9% of the use cases. That means: d) is out for me, for now, as is a)
* Option c): knowing all plots, but only present 1 at a time, makes simultaneous use of the different plots not easy, and creates a rather bodged/stateful interface.
* Option b): I'm not really happy with it, as it is rather wasteful in CPU time once you want the other plots, and the other plots are somewhat hidden. 

The challenge here is to transform the existing properties to keep it retro compatible, while at the same time not make it look bodged.

Therefore, I propose the following:
* Add an array of plots. These plots have all the relevant public properties and methods that the RawRead class has. Makes for easy client code adaptations. I can add preparations for the Trace class: data types, file positions, etc.
* add some other helper methods (example: list with all plot types)
* migrate the relevant properties to read-only properties 
* map the relevant properties (now read-only) to plot[0]
* clarify the proper use in the documentation

Although abstract classes will be helpful with the implementation of that, it will be rather artificial, as there is close to no difference in the structure or handling of the different plot types. What also is a problem is the lack of test material. I don't have enough test material, so going to deep in the plot type differentiation will be risky.

simonwaid, 2025-07-28:
I have attached a minimum example on how to create multiple plots. A typical use case would be to loop over multiple temperatures and/or operating voltates while doing op, ac and noise analysis under each condition. 

[plots_ex.txt](https://github.com/user-attachments/files/21467075/plots_ex.txt)

nunobrum, 2025-07-29:
@simonwaid I had quick look at the numpy.memmap. Also not sure it can be used here. The lazy loading would be a nice feature to have, the tricky part is to have it multi-thread friendly. This mean that each access would have to open and close the file, and we would need to have some kind of mutual exclusion on the file access, and the file structure (plots, variables, steps, etc...) would have to be stored in memory for fast access.

nunobrum, 2025-07-29:
> Just a heads up: added a load of test files, will now move the parser to a plot class, and adapt the properties, as proposed [above](https://github.com/nunobrum/spicelib/issues/218#issuecomment-3126219175)

I am OK with our proposal. Here is what I understood.
a) an Array of plots
b) Each Plot would have its own set of information (which must include the file offset). 
c) All read-only information is set as a properties

What I didn't understand was this: 
* map the relevant properties (now read-only) to plot[0]

nunobrum, 2025-07-29:
Can you give an example ?
From the example I saw, it seemed that each plot would have it's own attributes.

nunobrum, 2025-07-29:
OK! Now I understand what you mean. The confusion is that what you are implementing is what I called option c).
Option a) for me was that RawRead would be a function that would return Plots. One Plot if there was only one, a list of plots if there were many.
Anyway, I think your idea does solve more problems from the legacy point of view.

hb020, 2025-07-29:
The linked branch now holds working code for support of multiple sections in 1 file. 

TODO: 
* make headeronly work (it is ignored now)

hb020, 2025-07-30:
Should be good now. More robust, faster, more configurable.

closed 2025-07-31 by nunobrum