nunobrum/spicelib #222: Error in LTspice.create_list on Window 11

znsword, 2025-07-25:
Operating system
- Windows 11
- python 13.3 installed using uv
- spicelib installed using uv

When I run `spice_editor_example.py`,  there is an error in the LTspice.create_list.
```bash
uv run spice_editor_example.py
Traceback (most recent call last):
  File "E:\Users\Documents\Projects\eda\spice\test\spice_editor_example.py", line 4, in <module>
    LTspice.create_netlist('./testfiles/Noise.asc')
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Users\Documents\Projects\eda\spice\test\.venv\Lib\site-packages\spicelib\simulators\ltspice_simulator.py", line 351, in create_netlist        
    error = run_function(cmd_netlist, timeout=timeout, stdout=stdout, stderr=stderr)
  File "E:\Users\Documents\Projects\eda\spice\test\.venv\Lib\site-packages\spicelib\sim\simulator.py", line 37, in run_function
    result = subprocess.run(command, timeout=timeout, stdout=stdout, stderr=stderr)
  File "C:\Users\honestapple\AppData\Roaming\uv\python\cpython-3.13.3-windows-x86_64-none\Lib\subprocess.py", line 554, in run
    with Popen(*popenargs, **kwargs) as process:
        ~~~~~^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\honestapple\AppData\Roaming\uv\python\cpython-3.13.3-windows-x86_64-none\Lib\subprocess.py", line 1039, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                        pass_fds, cwd, env,
                        ^^^^^^^^^^^^^^^^^^^
    ...<5 lines>...
                        gid, gids, uid, umask,
                        ^^^^^^^^^^^^^^^^^^^^^^
                        start_new_session, process_group)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\honestapple\AppData\Roaming\uv\python\cpython-3.13.3-windows-x86_64-none\Lib\subprocess.py", line 1551, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                      ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
                            # no special security
                            ^^^^^^^^^^^^^^^^^^^^^
    ...<4 lines>...
                            cwd,
                            ^^^^
                            startupinfo)
                            ^^^^^^^^^^^^
FileNotFoundError: [WinError 2] 
```

So, I modified the `POpen class` in the subprocess.py according to [reference](https://blog.csdn.net/buling_me/article/details/125050817). The error info change to the following:

```bash
uv run spice_editor_example.py
'-netlist' is not an internal or external command, nor is it a runnable program or batch files.
Failed to create netlist
Traceback (most recent call last):
  File "E:\Users\Documents\Projects\eda\spice\test\spice_editor_example.py", line 4, in <module>       
    LTspice.create_netlist('./testfiles/Noise.asc')
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^
  File "E:\Users\Documents\Projects\eda\spice\test\.venv\Lib\site-packages\spicelib\simulators\ltspice_simulator.py", line 360, in create_netlist
    raise RuntimeError(msg)
RuntimeError: Failed to create netlist
```

What I want to do is to creat a netlist file with an asc file of LTSpice.

```python
asc_file_path = r'x:/xxxx/xxxx.asc'
netlist_file_path = LTspice.create_list(asc_file_path)
```
```

hb020, 2025-07-25:
where is your ltspice installed?

Can you run the following code and show the outcome?

```python
import logging
import spicelib
# set the logger to print to console and at debug level
loglevel = logging.DEBUG
logger = logging.getLogger(__name__)
logging.basicConfig(level=loglevel)
spicelib.set_log_level(loglevel)

from spicelib.simulators.ltspice_simulator import LTspice
print(f"Simulator is available: {LTspice.is_available()}")
print(f"Simulator exe: {LTspice.spice_exe}")
print(f"Simulator process name: {LTspice.process_name}")
LTspice.create_netlist("../bla.asc")  # <= substitute with the file name you have
```

hb020, 2025-07-25:
Have created a PR that will make it a bit more robust: will print a message if ltspice is missing, instead of just throwing strange error messages.

closed 2025-07-25 by nunobrum