nunobrum/spicelib #233: Keyboard package used by run_server.py requires root

simonwaid, 2025-08-08:
When using `run_server.py`, I get the error:  `ImportError: You must be root to use this library on linux.` This is a known issue of the keyboard package. Running  `run_server.py` as root is not recommendable. It would be better to use pynput instead of keyboard to avoid the error. See [Stackoverflow: Import error: you must be root](https://stackoverflow.com/questions/48796147/import-error-you-must-be-root) for more details.

hb020, 2025-08-08:
ok, will work on that, will take it together with #234, as the code affected is close.

hb020, 2025-08-08:
Proposition: 
Since that library is only used like this:
```python
while server.running():
    time.sleep(0.2)
    # Check whether a key was pressed
    if keyboard.is_pressed('q'):
        server.stop_server()
        break
```

I propose to stop using keyboard, and move to Ctrl-C handling. That is portable, and does not require root.

hb020, 2025-08-08:
Nice idea. Will test it, as we both use Threading and multiprocessing, and they are known to have some problems with KeyboardInterrupt. A central signal handler is sometimes safer.

nunobrum, 2025-08-10:
Integrated in 1.4.5

closed 2025-08-10 by nunobrum