nunobrum/spicelib #236: inconsistent formulas in montecarlo.py

TrainWrecka, 2025-08-08:
In the `montecarlo.py` file the function `prepare_testbench` uses a different formula for components and parameters if a deviation of the type DeviationType.minmax. is set. In theory I am expecting the lines 87 and 109 to both use 

```
new_val = "{urng(%s,%g,%g)}" % (val, (dev.max_val+dev.min_val)/2, (dev.max_val-dev.min_val)/2)
```

and line 90 and 112 

```
new_val = "{nrng(%s,%g,%g)}" % (val, (dev.max_val+dev.min_val)/2, (dev.max_val-dev.min_val)/6)
```

Otherwise, the monte carlo simulation uses the function `min_val+flat(max_val)` leading to an unexpected value range. Additionally, the issue reported in #235 results in the third parameter of the equations above to be negative, although this does not affect the simulation. Both issues together result in an awkward workaround to get the expected behavior of the `set_deviation` function in `tolerance_deviations.py`.

nunobrum, 2025-08-10:
Oh! Jeezz. That's right. At a given point I had updated the formula but, only updated the function call in one use case.
This will be corrected ASAP.

nunobrum, 2025-08-10:
Fixed in 1.4.5

closed 2025-08-10 by nunobrum