nunobrum/spicelib #246: Ngspice simulation runs but the raw file is not generated

NooriDan, 2025-08-19:
Hi,

the simulator does not save the raw file when I try to run ngspice on the circuit that I created in Xschem and then generated the ngspice netlist, (unless i explicitly mention the name of the raw file in the netlist and call the ngspice write method)

there is another issue that when I input the netlist via an instance of SpiceEditor the simulator complains that I have not defined a certain parameters (which I can confirm i have because i can run the ngspice manually). The solution i found is to just use the path to the netlist instead of passing the SpiceEditor object representation.

Files attached for your reference:

1. the gf180 nm model files [ngspice.zip](https://github.com/user-attachments/files/21848469/ngspice.zip)
2. the simple common source nmos amplifier schematic [cs_r_load_ac.spice.zip](https://github.com/user-attachments/files/21848474/cs_r_load_ac.spice.zip)

NooriDan, 2025-08-19:
PS: you'll need to updat the model files in the netlist 

[image]

and for your reference, I'm using spicelib in this very useful Docker image https://github.com/iic-jku/IIC-OSIC-TOOLS

hb020, 2025-08-19:
Looking at it, as ngspice indeed is a bit finicky with raw file writing. 
What version of spicelib is in the container image? There have been quite some changes in spicelib since the latest release of IIC-OSIC-TOOLS.

NooriDan, 2025-08-21:
-What version of spicelib is in the container image? There have been quite some changes in spicelib since the latest release of IIC-OSIC-TOOLS.? **im using their latest image on macos, i did notice their version was behind and it was giving errors (they had it on 1.4.4 and I updated it to 1.4.5)**

- Can you provide some of the python code you use? It would be hard to reproduce the problems without that? **Im using a simple jupyter notebook for now since I plan on using spicelib in optimizing larger scale Low Drop Out (LDO) design. I've attached the notebook to this reply **
[sizing_cs_signle_ended.ipynb](https://github.com/user-attachments/files/21909064/sizing_cs_signle_ended.ipynb)

- I see 4 ngspice files in your zip file, which one is the main file? **cs_r_load_ac.spice is the "main" file and you'd need the other 3 since they are the pdk file**
- what is the compatibility mode you have set via spicelib? That influences ngspice a lot. Normally spicelib forces ngbehavior=kiltspa? **I will need to figure that out, it probably is the default. i will let you know soon**

NooriDan, 2025-08-21:
I was curious if this issue is a known limitation?
> there is another issue that when I input the netlist via an instance of SpiceEditor the simulator complains that I have not defined a certain parameters (which I can confirm i have because i can run [...]

NooriDan, 2025-08-21:
Note: Compatibility modes selected: hs a

hb020, 2025-08-21:
So the raw file writing is OK now, but I will make it more clear in the readme. 

> I was curious if this issue is a known limitation?
> there is another issue that when I input the netlist via an instance of SpiceEditor the simulator complains that I have not defined a certain [...]

Not a limitation I know of. Will look into it.

hb020, 2025-08-21:
The correction in the PR solves the "file or editor" problem.

Also, you should use 
```
ngspice_simulator.NGspiceSimulator.set_compatibility_mode("hsa")

runner = SimRunner(
    simulator=ngspice_simulator.NGspiceSimulator.....
``` 
because otherwise it will run in LTspice compatibility mode, and choke on your lib annotations ('typical', 'res_typical').