nunobrum/spicelib #254: '+' After line breaks not handled correctly

simonwaid, 2025-09-30:
In Spice, a '+' after a line break means the next line should be added to the previous line. E.g. 

```
x21 VP In_A vin_a vguard_a VN pad_in_guard
x41 Res_2 VP VP2 v_fb outint_a Res vin_a Out_A Pulse_In vguard_a Pulse_En_B Bias_Out2 V_ref bias_a1 bias_a2 bias_ar VNEG_A VN
+ CSA_varC
x63 VP bias_a1 bias_a2 bias_ar VN Curref_glob_one

```

After loading the above netlist with the SpiceLib editor and writing it to the disk, the '+' is missing:

```
x21 VP In_A vin_a vguard_a VN pad_in_guard
x41 Res_2 VP VP2 v_fb outint_a Res vin_a Out_A Pulse_In vguard_a Pulse_En_B Bias_Out2 V_ref bias_a1 bias_a2 bias_ar VNEG_A VN
 CSA_varC
x63 VP bias_a1 bias_a2 bias_ar VN Curref_glob_one
```

hb020, 2025-09-30:
will look into it, this might be a regression since the last changes.

hb020, 2025-09-30:
Hi, are you sure to have the latest version from git? Are you using Windows?

When I have a netlist with the content you gave, and then save (on MacOS), I get this:
```text
x21 VP In_A vin_a vguard_a VN pad_in_guard
x41 Res_2 VP VP2 v_fb outint_a Res vin_a Out_A Pulse_In vguard_a Pulse_En_B Bias_Out2 V_ref bias_a1 bias_a2 bias_ar VNEG_A VN CSA_varC
x63 VP bias_a1 bias_a2 bias_ar VN Curref_glob_one
```
Only 3 lines.